cu-ecen-aeld/final-project-akshita-bhasin #9: Restructure github code to integrate various peripheral functionalities into a single executable application

jajoosiddhant, 2020-04-16:
https://github.com/cu-ecen-5013/final-project-akshita-bhasin/blob/b2668a19d1bb75221d1f6eab1b8d77624f68d8b0/test_bbb/uart/uart.c#L13

@akshita-bhasin 
I can see you have created separate executables for different peripherals. 
I would advice you to have a single executable i.e a single main and have different peripheral header files to be included in main.c leading up to only a single executable to deal with. 
As far as testing those peripherals is concerned, you can create a test function in the respective source file and call that particular function in main.c to check whether your peripheral is functioning as expected. 
Your current method would lead you to invest a lot more time at the end to integrate all the source files in a single application and create an application out of it. Start creating APIs as and when you try to make the peripheral work so that you can just reuse it anywhere in the application by just including the header file.
As the project progresses you would require dependencies of several source files that you create in other source files. So you would eventually have to restructure your github too for that.
Currently there is no initialization function, no read and write function for UART peripheral as well. Just a plain simple testing code with the entire code in main().
Having a testing function(eg: uart_loopback_test()) would also help you to figure out if the peripheral is working or not if you get stuck at some point in your project to debug the application.

jajoosiddhant, 2020-04-16:
You can simply have two folders with the name src and inc with all your source and header files in it with a main.c outside calling functions from those source files.

jajoosiddhant, 2020-04-16:
@madhukararora Please note the above instructions for Freertos code as well. Have a separate uart.c and uart.h file for all the functions related to UART.

akshita-bhasin, 2020-04-16:
Started with integrating UART, done successfully and tested, need to add shared memory implementation when testing with TMP102 as well.

akshita-bhasin, 2020-04-19:
Restructured code to add temperature sensor task, uart task and actuators task on BBB application side.
Commit for [Application source](https://github.com/cu-ecen-5013/final-project-akshita-bhasin/commit/4fd7fef2ea22016eec09701d44f17c233e2bd270)
Commit for [Buildroot update with Actuators intergated](https://github.com/cu-ecen-5013/final-project-madhukararora/commit/a0de98cd73a4617287fb5fdc1d3e066fdfd16c0e)

akshita-bhasin, 2020-04-19:
Updating FreeRTOS side of the code next.

akshita-bhasin, 2020-04-19:
FreeRTOS restructuring done.
Commit for [FreeRTOS Source code](https://github.com/cu-ecen-5013/s20-environmental-monitoring-tiva/commit/a1acb1c2fbe1a1d89136b88375f8b28d3ceeb435)

akshita-bhasin, 2020-04-19:
Application on BBB
[image]

Application on FreeRTOS
[image]

closed 2020-04-19 by akshita-bhasin